project-march/hardware-interface #30: Refactor settings to use constants instead of memory adresses

Ishadijcks, 2019-04-03:
Create a header file with constants with something like this:

```
MODE_OF_OPERATION_ADDRESS = 0x1234
```

And refactor these statements with the consts.

```
success &= sdo_bit8(slaveIndex, 0x6060, 0, 8);
```

Ishadijcks, 2019-04-30:
@martijnvandermarel I'm a bit conflicted on how to implement this cleanly. It would make sense to add them as IMCObjects, but that would be odd in the PDOmapping, thoughts?

martijnvandermarel, 2019-04-30:
We could make a seperate IMC addresses constants file, and refactor PDOmapping with these constants as well, alongside the sdo writing instances

Ishadijcks, 2019-04-30:
Hmm that would require you to add a new PDOobject in 3 places, that's kind of overkill. Will probably need to refactor the IMCObject enum to include the address.

Roelemans, 2019-09-24:
Seems to be finished to some extend; will not look into it untill we discover the problem ourselves.

closed 2019-09-24 by Roelemans