project-march/hardware-interface #48: Only check for software limits when a joint can actuate

Ishadijcks, 2019-05-23:
When the hardware_interface gets launched, there is a check to see if we are within safetylimits [here](https://github.com/project-march/hardware-interface/blob/develop/march_hardware_interface/src/march_hardware_interface.cpp#L95).

However sometimes while debugging the encoder, this check crashes because the encoder will read a 0, which is almost always out of its limits.

@martijnvandermarel @TimBuckers @Bart-van-Ingen, I'd like each of your opinions on this issue. Should it be possible to allow continuation when *outside the limits while actuationmode is off*?

Ishadijcks, 2019-05-23:
When the encoder returns 0, you want to move it and see if the value changes. This is the most common usecase I believe

JitskedeVries, 2019-05-23:
Do you mean move the joint by hand and see if the value changes? I think that must be allowed.

Actuating the motor to see if the value changes, must not be allowed. You could obviously break something if we don't know what the encoder value is and we don't know to what value we can move.

JitskedeVries, 2019-05-23:
I agree with you both. The encoder can be outside its limits when launching, especially when you want to test something else. But when you are going to actuate, the encoder value must be inside it limits. Thus the error can be thrown later in the process (just before actuating).

Ishadijcks, 2019-05-23:
This would be based off the `allowActuation` property which cannot change at runtime.

PR #70 (2019-06-25): only throw exception when allow actuation is true
closes #48
closed 2019-06-25 by Ishadijcks