project-march/hardware-interface #113: Fix undervoltage error IMC

JitskedeVries, 2019-07-23:
The IMC's keep throwing an undervoltage error, which looks like this.

``` 
[ WARN] [1563866879.067607010]: EtherCAT rate of 4 milliseconds per cycle was not achieved for 3.640000 percent of all cycles
[ WARN] [1563866884.920797020]: All-High-Voltage disabled
terminate called after throwing an instance of 'std::runtime_error'
 what():  IMotionCube of joint right_ankle is in fault state Fault
Detailed Error: (0000000000000000)
Motion Error: Under-voltage. (0010000000000000)
```

Either the High Voltage is disabled by the PDB because of a weird reason or the voltage drops down because of something else. Either way it should be fixed, since it costs a lot of time and it is dangerous when Sjaan is in the exoskeleton.

Ishadijcks, 2019-07-23:
When does this occur? Always? Startup? Specific gait?

Can you provide a bag file in which this occurs?

Can you add more of the log?

martijnvandermarel, 2019-07-23:
I have increased the timeout parameter in the PDB software which is used to check if the master is still online, which I think should fix this issue.

If the PDB hasn't received a signal from the master for a certain amount of time (was 100 ms, now 1 s) HV is disabled. Since you note that you get a warning that HV is disabled, this is most likely the cause.

I think this issue hasn't been seen with the standalone PDB software, which further supports my theory.

I've updated the PDB with the above would-be-fix, please let me know if the bug still occurs.

Branch with fix:
https://github.com/project-march/ethercat-slaves/tree/fix/increase-masteronline-time

JitskedeVries, 2019-07-23:
> When does this occur? Always? Startup? Specific gait?
This occurs during runtime, no specific gait. It mostly happens while we are not running any gait. But I think that is a coincidence.

> Can you provide a bag file in which this occurs?
I'll come back to this. 

> Can you add more of the log?
``` [ WARN] [1563866816.357339452]: EtherCAT rate of 4 milliseconds per cycle was not achieved for 1.120000 percent of all cycles
[ WARN] [1563866826.845216149]: EtherCAT rate of 4 milliseconds per cycle was not achieved for 2.240000 percent of all cycles
[ WARN] [1563866837.230408877]: EtherCAT rate of 4 milliseconds per cycle was not achieved for 2.240000 percent of all cycles
[ WARN] [1563866847.706721979]: EtherCAT rate of 4 milliseconds per cycle was not achieved for 2.320000 percent of all cycles
[ WARN] [1563866858.171276550]: EtherCAT rate of 4 milliseconds per cycle was not achieved for 2.720000 percent of all cycles
[ WARN] [1563866879.067607010]: EtherCAT rate of 4 milliseconds per cycle was not achieved for 3.640000 percent of all cycles
[ WARN] [1563866884.920797020]: All-High-Voltage disabled
terminate called after throwing an instance of 'std::runtime_error'
 what():  IMotionCube of joint right_ankle is in fault state Fault
Detailed Error: (0000000000000000)
Motion Error: Under-voltage. (0010000000000000)
[march_hardware_interface-3] process has died [pid 7650, exit code -6, cmd /home/march/Documents/march_ws/devel/lib/march_hardware_interface/march_hardware_interface_node march4 __name:=march_hardware_interface __log:=/home/march/.ros/log/20b6ee40-ad1b-11e9-a955-5c5f67b4a253/march_hardware_interface-3.log].
log file: /home/march/.ros/log/20b6ee40-ad1b-11e9-a955-5c5f67b4a253/march_hardware_interface-3*.log
```

closed 2019-07-29 by martijnvandermarel