project-march/hardware-interface #128: Memory leak

JitskedeVries, 2019-07-31:
While airgaiting with the we got the following error a couple of times: 
`*** Error in /home/march/Documents/march_ws/devel/lib/march_hardware_interface/march_hardware_interface_node': malloc(): memory corruption: 0x00007f0a78001190 *** `

We launched the march_hardware_interface march4.launch at the feature/torquemode branch. But I suspect that this could also be a thing on develop. It might have something to do that we read and write to the same memory location at the same time.

The complete error as published in the terminal:
[error.txt](https://github.com/project-march/hardware-interface/files/3450417/error.txt)

Ishadijcks, 2019-07-31:
(_ZN5boost6detail11thread_dataINS_3_bi6bind_tIvNS_4_mfi3mf0IvN14realtime_tools17RealtimePublisherIN22 *march_shared_resources23AfterLimitJointCommand* _ISaIvEEEEEEENS2_5list1INS2_5valueIPSC_EEEEEEE3runE

Where is this file hosted?

Olavhaasie, 2019-09-24:
Not reproducible

closed 2019-09-24 by Olavhaasie