project-march/hardware-interface #142: Debugger

JitskedeVries, 2019-08-28:
For debugging it is much easier if we have a debugging tool for the hardware interface. 

We could not get the clion debugger to work with the hardware interface, since there are functions which depend on variables within a certain time frame. Therefore the hardware interface could not be stopped at stop points.

Also it might be necessary to use the debugger in the ROS libraries, since those package are not perfect.

Ishadijcks, 2019-08-28:
The `hardware_interface` uses SOEM, which simply cannot be debugged in real-time.

What exactly is the goal of this issue, and when can it ever be closed?

Ishadijcks, 2019-08-28:
Minor correction, the CLion debugger attachers a GDB debugger to your process, it does not have its own debugger.

Roelemans, 2019-09-24:
Does not seem possible.

closed 2019-09-24 by Roelemans