project-march/hardware-interface #149: Difference between target and actual also handled in gait scheduler/controller

TimBuckers, 2019-09-01:
The controller already throws an error when the difference between path (target) and actual is too big. This is no handled by the scheduler node, by freezing (see https://github.com/project-march/march-iv/pull/340). However, in position mode this check is also done in the hardware-interface. This is handled by crashing the hardware interface (not freezing).

Possible solutions:
- [ ] remove this check entirely from hardware interface. This is not perfect, because this check adds safety, however this makes the behavior more consistent between position and torque mode.
- [ ] make the check in the controller more restrict than the hardware interface check (this means freeze first and crash only when it becomes worse). So probably make the 0.27 rad in the hardware interface larger not the 0.27 smaller in the controller, because the 0.27 is already causing many crashes on trainings.

martijnvandermarel, 2019-09-01:
I would go for the second one for the next training and see if it works as intended. Seems relatively easy

JitskedeVries, 2019-09-01:
I agree with Martijn. We shouldn't remove the extra safety. (Altough in Torque Mode there is no extra safety.)

Olavhaasie, 2019-09-24:
This has been fixed in #150

closed 2019-09-24 by Olavhaasie